STMicroelectronics/stm32f1xx-hal-driver #1: extern uwTick : Missing from F1 variant stm32f1xx_hal.h

geosmall, 2020-05-14:
Issue raised here https://github.com/STMicroelectronics/STM32CubeF1/issues/12

Issue was first raised here https://github.com/stm32duino/Arduino_Core_STM32/issues/1064

**Describe the bug** :

```extern __IO uint32_t uwTick;```

This declaration is missing from F1 variant, present on all other variants.

https://github.com/STMicroelectronics/STM32CubeF1/blob/5ddefcd6a64e793087436cc4ed42abd1886952c1/Drivers/STM32F1xx_HAL_Driver/Inc/stm32f1xx_hal.h#L59-L61

**To Reproduce** :

By inspection

See as example:
https://github.com/STMicroelectronics/STM32CubeF4/blob/5d01400afd60410f6e049cbd19179a67d44d53fd/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal.h#L200-L207
and
https://github.com/STMicroelectronics/STM32CubeF7/blob/08376dce1b404687e4a86b73077f396bccfc9cb5/Drivers/STM32F7xx_HAL_Driver/Inc/stm32f7xx_hal.h#L195-L201

ASELSTM, 2020-05-15:
Hi @geosmall,

Thank you for having pointed out this issue.

I have noticed that this issue has also been posted into the **STM32CubeF1** firmware repository. Actually, having all issues centralized there would greatly ease for us tracking them. Hence, please allow me to mark this one as a duplicate of the other and to close it.

With regards,

ASELSTM, 2020-05-15:
Duplicate of  #https://github.com/STMicroelectronics/STM32CubeF1/issues/12

ASELSTM, 2020-05-15:
#https://github.com/STMicroelectronics/STM32CubeF1/issues/12

closed 2020-05-15 by ASELSTM

geosmall, 2020-05-16:
Thanks.  I wasn't sure where best to raise the issue.  I linked here also as the code is replicated.