STMicroelectronics/stm32f1xx-hal-driver #3: SetPinMode and SetPinPull sequence

mgiaco, 2020-10-08:
Hello,
I switched to the latest cubef1 today. Unfortunately, the device stopped working :-( 
I then found out that it had to do with these changes:

old:

```
     /* Pin Mode configuration */
      LL_GPIO_SetPinMode(GPIOx, currentpin, GPIO_InitStruct->Mode);

      /* Pull-up Pull-down resistor configuration*/
      LL_GPIO_SetPinPull(GPIOx, currentpin, GPIO_InitStruct->Pull);

      if ((GPIO_InitStruct->Mode == LL_GPIO_MODE_OUTPUT) || (GPIO_InitStruct->Mode == LL_GPIO_MODE_ALTERNATE))
      {
        /* Check speed and Output mode parameters */
        assert_param(IS_LL_GPIO_SPEED(GPIO_InitStruct->Speed));
        assert_param(IS_LL_GPIO_OUTPUT_TYPE(GPIO_InitStruct->OutputType));

        /* Speed mode configuration */
        LL_GPIO_SetPinSpeed(GPIOx, currentpin, GPIO_InitStruct->Speed);

        /* Output mode configuration*/
        LL_GPIO_SetPinOutputType(GPIOx, currentpin, GPIO_InitStruct->OutputType);
      }
```

new:
```
      if ((GPIO_InitStruct->Mode == LL_GPIO_MODE_OUTPUT) || (GPIO_InitStruct->Mode == LL_GPIO_MODE_ALTERNATE))
      {
        /* Check speed and Output mode parameters */
        assert_param(IS_LL_GPIO_SPEED(GPIO_InitStruct->Speed));
        assert_param(IS_LL_GPIO_OUTPUT_TYPE(GPIO_InitStruct->OutputType));

        /* Speed mode configuration */
        LL_GPIO_SetPinSpeed(GPIOx, currentpin, GPIO_InitStruct->Speed);

        /* Output mode configuration*/
        LL_GPIO_SetPinOutputType(GPIOx, currentpin, GPIO_InitStruct->OutputType);
      }

      /* Pull-up Pull-down resistor configuration*/
      LL_GPIO_SetPinPull(GPIOx, currentpin, GPIO_InitStruct->Pull);

      /* Pin Mode configuration */
      LL_GPIO_SetPinMode(GPIOx, currentpin, GPIO_InitStruct->Mode);
```
Why was this changed? I didn´t found any issue for that in this repo. So i also found an issue here  https://community.st.com/s/global-search/LL_GPIO_Init seems to be a little bit related. Can someone have a look on that?

many thanks

ASELSTM, 2020-10-08:
Hi @mgiaco,

First, we would like to thank you for your contribution.

Actually, this change aims to correct the GPIO initialization sequence to avoid unwanted pulse on GPIO Pin’s. Indeed, old configuration sequence

```

1. Pin Mode configuration
2. Pull-up Pull-down resistor configuration
3. Speed mode configuration
4. Output mode configuration

```
causes unwanted pulse generation when initializing of the GPIO registers leading to an incorrect behavior. This sequence has therefore been corrected to avoid this case. The new sequence is then such as updated in the latest release.

```

1. Speed mode configuration
2. Output mode configuration
3. Pull-up Pull-down resistor configuration
4. Pin Mode configuration

```
 
The fix was also mentioned in the [Release Note](https://htmlpreview.github.io/?https://github.com/STMicroelectronics/stm32f1xx_hal_driver/blob/master/Release_Notes.html) under **"HAL/LL GPIO driver"** section as following : 

> Update LL GPIO initialization sequence to avoid unwanted pulse on GPIO Pin’s

Hope this has given you a better understanding of the fix.

With regards,

mgiaco, 2020-10-12:
Okay so i have to check what´s wrong with my implementaiont.
many thanks

closed 2020-10-12 by mgiaco