STMicroelectronics/stm32f1xx-hal-driver #4: A bug when use HAL_UARTEx_ReceiveToIdle_DMA

rzzuozuo, 2021-09-15:
**Describe the bug**
Cannot distinguish between half complete in DMA and idle interrupts

**How To Reproduce**
1. set a rxSize

2. use HAL_UARTEx_ReceiveToIdle_DMA

3. send half the data （rxSize/2）

4. HAL_UARTEx_RxEventCallback will be called 2 times with the same input parameter(size)

KORKAD, 2021-09-22:
Hello @rzzuozuo

This is the normal behavior for this USART service:

     Use of HAL_UARTEx_ReceiveToIdle_DMA service, will generate calls to
     user defined HAL_UARTEx_RxEventCallback callback for each occurrence of
     following events :
     - DMA RX Half Transfer event (HT)
     - DMA RX Transfer Complete event (TC)
     - IDLE event on UART Rx line (indicating a pause is UART reception flow)

for more details please refer to the [UART_ReceptionToIdle_CircularDMA](https://github.com/STMicroelectronics/STM32CubeF4/tree/master/Projects/STM32446E-Nucleo/Examples/UART/UART_ReceptionToIdle_CircularDMA) example available on STM32CubeF4 package.

With Regards

ASELSTM, 2021-11-30:
Hi @rzzuozuo,

Please allow me to close this thread as answered. Thank you for your comprehension and thank you once more for your contribution.

With regards,

closed 2021-11-30 by ASELSTM