STMicroelectronics/stm32f1xx-hal-driver #6: Bug in HAL_FLASHEx_Erase_IT - if process already ongoing then pFlash remains locked

pi, 2023-07-24:
<pre>
HAL_StatusTypeDef HAL_FLASHEx_Erase_IT(FLASH_EraseInitTypeDef *pEraseInit)
{
  HAL_StatusTypeDef status = HAL_OK;

  /* Process Locked */
  __HAL_LOCK(&pFlash);

  /* If procedure already ongoing, reject the next one */
  if (pFlash.ProcedureOnGoing != FLASH_PROC_NONE)
  {
>>> here must be  __HAL_UNLOCK(&pFlash); <<<<
    return HAL_ERROR;
  }
</pre>

RJMSTM, 2023-07-28:
Hello @pi,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,
Rania

ALABSTM, 2023-08-11:
See also [STM32CubeF4#88](https://github.com/STMicroelectronics/STM32CubeF4/issues/88).

TOUNSTM, 2023-08-18:
Hi @pi,

Could you please give us more details on how you encountered this issue so that we can better analyze it?

With regards,

pi, 2023-09-01:
Hi @TOUNSTM

I found this error while examining the function code. An error can occur when calling this function from different tasks without additional synchronization of access to flash memory.

TOUNSTM, 2024-06-26:
Hello @pi,

Indeed, the **LOCK/UNLOCK** mechanism is set to be removed from the **HAL_FLASHEx_Erase_IT()** and **HAL_FLASH_Program_IT()** APIs. Consequently, your change request is no longer applicable. Please allow me to close the issue.

Thank you for your understanding, and we appreciate your report.

TOUNSTM, 2024-06-26:
ST Internal Reference: 185075

closed 2024-06-26 by TOUNSTM