STMicroelectronics/stm32f1xx-hal-driver #8: Changing the write-only AFIO_MAPR.SWJ_CFG field in LL

wek3, 2024-02-17:
In stm32f1xx_ll_gpio.h, LL_GPIO_AF_EnableRemap_SWJ(), LL_GPIO_AF_Remap_SWJ_NONJTRST, LL_GPIO_AF_Remap_SWJ_NOJTAG(), LL_GPIO_AF_DisableRemap_SWJ() all use

`SET_BIT(AFIO->MAPR, AFIO_MAPR_SWJ_CFG_xxxxx);`

which is incorrect, as SET_BIT is

`#define SET_BIT(REG, BIT) ((REG) |= (BIT))`

and the read value of AFIO_MAPR.SWJ_CFG field according to RM0008 is unpredictable.

This has been reported in https://community.st.com/t5/stm32-mcus-embedded-software/incorrectly-written-afio-mapr-swj-cfg-bits-in-cubef1-ll/m-p/90756 , can you please drop a comment there, too.

Thanks,

Jan Waclawek

TOUNSTM, 2024-02-19:
Hello @wek3,

Thank you for your contribution. This issue has been fixed in the frame of version v1.1.9 of the stm32f1xx_hal_driver. Please allow me then to close this thread.

With regards,

closed 2024-02-19 by TOUNSTM

wek3, 2024-02-19:
Thanks, @TOUNSTM .
JW