STMicroelectronics/stm32f1xx-hal-driver #9: Interrupting USB Transfer

AndyD87, 2024-02-28:
https://github.com/STMicroelectronics/stm32f1xx_hal_driver/blob/ddf259a6eb9dc4ba421f5e2b1ae619bd77468214/Inc/stm32f1xx_hal_pcd.h#L868

This line of code is interrupting my USB Transfer in random time on STM32F103.
In Wireshark i see the OUT Endpoint answered with USBD_STATUS_STALL_PID

The host is continously sending Data to the OUT Pipe.
While debugging the **HAL_PCD_EP_Receive** , until this line of code, the connection is stable.
Stepping over it will interrupt the Transaction and on Host the Pipe is broken.

In normal running, the next line in Debug, will reactivate the Pipe to valid state, so the host is not recognizing it.
But some interrupts occour within this two lines and the transfer is broken.
The two lines are part of **USB_EPStartXfer**:
 - 2441: PCD_SET_EP_RX_CNT(USBx, ep->num, len);
 - 2467: PCD_SET_EP_RX_STATUS(USBx, ep->num, USB_EP_RX_VALID);

After deleting this line, i had no transfer interruptions any more.

AndyD87, 2024-02-28:
The 0x3FFU is generating an PMA Overrun wich is resulting in an stalled OUT Endpoint.

The Host is running a WinUSB Driver wich could not handle this Error State.
A pipe reset was required to continue the transfer.

The STM32F1 is running wihtout any peripherals, just USB and the HSE Oscilator was activated with CubeMX.
The USB Part was configured with 1 OUT and 1 IN EP with 64 Byte of maximum packet size for Bulktransfer.
The STM32F1 Application was reduced to a while(1) which is coninously calling HAL_PCD_EP_Receive and a short timeout of 1ms to simulate the data handling.
The Error happend on different devices.

ALABSTM, 2024-03-04:
Hi @AndyD87,

Thank you for all these details. Your report has been forwarded to our development teams. I will keep you informed.

With regards,

ALABSTM, 2024-03-04:
ST Internal Reference: 175167

AndyD87, 2024-03-04:
Thank you very much.
If you need support, just ask. 
I have saved the example project to reproduce the error.

ALABSTM, 2024-03-05:
Hi @AndyD87,

Thank you very much to propose your support. I'll let our development teams know you can provide a modified example allowing to reproduce the issue in case they need it. I will keep you informed.

With regards,

ALABSTM, 2024-04-15:
Hi @AndyD87,

I hope you are fine. Please excuse this delayed reply. Could you please share the modified example you mentioned? It would be helpful to us.

Many thanks,

ALABSTM, 2024-06-11:
Hi @AndyD87,

Replying to what you posted
>> The STM32F1 Application was reduced to a while(1) which is coninously calling HAL_PCD_EP_Receive and a short timeout of 1ms to simulate the data handling. 

Our development teams said this is not the way to program a USB transfer as this will necessarily break the ongoing transfer. Instead you should check that the current transfer has been completed (using the one the below callback , for instance) before submitting any new Rx transfer. A simple `while(1)` loop cannot allow that.

https://github.com/STMicroelectronics/stm32f1xx_hal_driver/blob/e0654977ea5d79a4bee5f153f62d53bf2e376602/Src/stm32f1xx_hal_pcd.c#L2372-L2377

I hope this gives you a clue on how to handle your application. Please allow me to close this issue. Thank you for your comprehension.

With regards,

closed 2024-06-11 by ALABSTM