STMicroelectronics/stm32f1xx-hal-driver #11: STM32F1xx RTC BKP Registers 16bit, but HAL requests 32bits

RedEchidnaUK, 2024-04-03:
I'm fairly new to MCU development and working with the STM32F1xx series, but I've been pointed over to here from the STM32duino project as they pull the HAL from here.

The RTC BKP registers on the STM32F103 (and I believe all of the STM32F1xx series) are 16bit, but the HAL driver (Inc/stm32f1xx_ll_rtc.h) requests and accepts an unsigned 32bit value. Of course it can't actually save 32bits only 16bits, so you lose the top 16bits of the data you sent. 

Obviously it is possible to work around this by splitting the data over 2 registers, but is this by design? Perhaps there is an MCU in the series that has a 32bit backup register and I've missed it? In my opinion, if all MCUs in the series have 16bit RTC BKP registers, the HAL should request and accept just 16bits to avoid confusion and accidental data loss.

ALABSTM, 2024-04-08:
Hi @RedEchidnaUK,

Thank you for this proposal. However, looking at page **86** of [RM0008](https://www.st.com/content/ccc/resource/technical/document/reference_manual/59/b9/ba/7f/11/af/43/d5/CD00171190.pdf/files/CD00171190.pdf/jcr:content/translations/en.CD00171190.pdf), revision 21, you would notice that the backup registers are actually **32-bit wide**, with the 16 MSB being **reserved** (screenshot below).

[image]

These 16 reserved bits are simply **not** represented on page **83** (screenshot below).

[image]

Now, regarding the accidental data loss, I guess you mean storing a 32-bit wide data in one of the backup registers and, thus, having the 16 MSB lost as coinciding with the reserved part of the register. In this case, it is up to the user to pay attention.

One last comment, if I may.
* As the STM32F1xx microcontrollers are 32-bit microcontrollers, data is "manipulated" in "chunks" of 32 bits.
  * This data should hence be stored in memory (particularly RAM), at addresses **multiple of** 32 bits or 4 bytes (**e.g.**, 0x2000'0000, 0x2000'0004).
  * This is what we call data **alignment** in memory.
* In case of 16-bit wide data, it would be stored at addresses 0x2000'0000 and 0x2000'0002.
  * Supposing the CPU of a 32-bit microcontroller would fetch the data at address 0x2000'0000, the data at address 0x2000'0002 would be fetched along with it and **extra operations** would be needed to separate both. This would result in a **loss of performance**.
  * Moreover, if the CPU would fetch the data at address 0x2000'0002, it would be unable to do it as this would yield a **fault** due to an **unaligned** memory access. 

I hope this helps. I let you dig deeper into this topic. Thank you again for your proposal. Please allow me to close this issue.

With regards,

closed 2024-04-08 by ALABSTM

RedEchidnaUK, 2024-04-09:
Thank you for the detailed explanation.