STMicroelectronics/stm32f1xx-hal-driver #13: extraneous parentheses

RetoFx, 2024-05-26:
Hi
Mi ARMCLang report in file stm32f1xx_hal_usart.c Line 2785

if((husart->Instance == USART1))

../../HAL/STM32/STM32F1xx/stm32f1xx_hal_usart.c(2785): warning: equality comparison with extraneous parentheses [-Wparentheses-equality]
  if((husart->Instance == USART1))
      ~~~~~~~~~~~~~~~~~^~~~~~~~~
../../HAL/STM32/STM32F1xx/stm32f1xx_hal_usart.c(2785): note: remove extraneous parentheses around the comparison to silence this warning
  if((husart->Instance == USART1))
     ~                 ^        ~
../../HAL/STM32/STM32F1xx/stm32f1xx_hal_usart.c(2785): note: use '=' to turn this equality comparison into an assignment

Please replace it with 

if(husart->Instance == USART1)

ASEHSTM, 2024-05-31:
Hello @RetoFx,

Thank you for your contribution. This will be fixed soon.

With regards,

ASEHSTM, 2024-05-31:
ST Internal Reference: 182974

TOUNSTM, 2024-06-28:
fixed in 6ca7fc40cc862623dc05b33b0014789c1fbbbb86

closed 2024-06-28 by TOUNSTM