STMicroelectronics/stm32f1xx-hal-driver #14: In HAL_UART_IRQHandler function, I found a small issue. Please take a look and confirm it.

qulei123, 2025-03-10:
In stm32f1xx_hal_uart.c, I found a small issue. Please take a look and confirm it.

version：v1.1.10

At line 2481 of the HAL_UART_IRQHandler function.
```
void HAL_UART_IRQHandler(UART_HandleTypeDef *huart)
{
    ....
  /* Check current reception Mode :
     If Reception till IDLE event has been selected : */
  if ((huart->ReceptionType == HAL_UART_RECEPTION_TOIDLE)
      && ((isrflags & USART_SR_IDLE) != 0U)
      && ((cr1its & USART_SR_IDLE) != 0U))
    ....
}
```
Whether it should be modified to: 
```
  /* Check current reception Mode :
     If Reception till IDLE event has been selected : */
  if (  (huart->ReceptionType == HAL_UART_RECEPTION_TOIDLE)
      &&((isrflags & USART_SR_IDLE) != 0U)
      &&((cr1its & USART_CR1_IDLEIE) != 0U))
```

KRASTM, 2025-03-13:
ST Internal Reference: 205308

KRASTM, 2025-04-08:
Fixed in Commit: https://github.com/STMicroelectronics/stm32f1xx-hal-driver/commit/38d14024a9d3f6802506c3ec7a4308563760e54e

closed 2025-04-08 by KRASTM